LowPowerLab/RFM69 #202: Default 433 MHz frequency is illegal

awjlogan, 2025-07-31:
([cross post from rpi-rfm69 library](https://github.com/jgillula/rpi-rfm69/issues/36#issue-3280277491), direct port of LPL)

The default frequency for the 433 MHz system is 433.00 MHz. However, this is illegal as the ISM 433 MHz band is defined from 433.05 to 434.79 MHz, with a centre frequency of 433.92 MHz. Correspondingly, *all* of the lower sidebands and at least the first upper sideband are outside the designated ISM band when using this value.

The suggested change is to set:

- RF_FRFMID_433 to 0x7A
- RF_FRFLSB_433 to 0xE1

This puts the centre frequency at exactly 433.92 MHz in accordance with the regulations.

Unfortunately this is a breaking change - I've tried asymmetric RFM pairs (one at 433.00 MHz and the other at 433.92 MHz) and there's no transmission. Setting this as the default would likely break a lot of systems, so perhaps this is best as a feature flag that is set. Happy to open a PR if desired.

This has also been [discussed elsewhere](https://forum.mysensors.org/topic/6487/rfm69-433mhz-ism-band).

closed 2025-10-17 by LowPowerLab

awjlogan, 2025-10-17:
Thanks @LowPowerLab - and no worries, you are not obliged to do anything in anyone's time but your own! Might be worth a note somewhere that this is incompatible with existing setups, but hopefully this issue should cover it. Hope all's well otherwise.

LowPowerLab, 2025-10-20:
@awjlogan Thanks for the understanding and patience. I made a [blog post](https://lowpowerlab.com/2025/10/17/rfm69-v1-6-breaking-change-433-0mhz-is-now-433-92mhz/) to document this and also I thought this and other recent commits deserved a [new library release](https://github.com/LowPowerLab/RFM69/releases/tag/1.6.0).